MyoHub/myosuite #296: RL-E2E performance betther than reported.

InckieUltra, 2025-03-31:
Hi, myo team.

I'm currently playing with myosim models and reproducing SAR and DynSyn (https://github.com/Beanpow/DynSyn). The two papers both use some techniques to reduce the action space dimension.

When reproducing the RL-E2E results on myoLegWalk-v0 environments (code from DynSyn), I found that the RL-E2E baseline can easily reach a reward of 1e4 within 3e6 timesteps. And the result reported in SAR is 2k reward/3e6 timesteps, the result reported in DynSyn is about 3~4k reward/3e6 timesteps.

So, to my surprise, RL-E2E's performance seems to be too good.

Here is a minimal example of using stable baselines 3 SAC  to deal with myoLegWalk-v0, which is similar to the implementation in DynSyn E2E baseline:

`import tqdm
from myosuite.utils import gym
from stable_baselines3 import SAC
from stable_baselines3.common.vec_env import SubprocVecEnv, VecNormalize, DummyVecEnv
from stable_baselines3.common.monitor import Monitor
from docs.source.tutorials.SAR.SAR_tutorial_utils import SaveSuccesses, linear_schedule, configure
import os
import imageio
from PIL import Image

def make_env(env_id):
    def _init():
        env = gym.make(env_id)
        env = Monitor(env)
        return env
    return _init

def train(env_name, policy_name, timesteps, seed):
    env = SubprocVecEnv([make_env(env_name) for _ in range(16)])
    env = VecNormalize(env, norm_obs=True, norm_reward=False, clip_obs=10.0)
    
    net_shape = [256, 256]
    policy_kwargs = dict(net_arch=dict(pi=net_shape, qf=net_shape))
    
    model = SAC('MlpPolicy', env, learning_rate=linear_schedule(.001), buffer_size=int(1e6), train_freq=(4, 'step'),
            gradient_steps=-1, policy_kwargs=policy_kwargs, verbose=1)
    
    succ_callback = SaveSuccesses(check_freq=1, env_name=env_name+'_'+str(seed), 
                             log_dir=f'{policy_name}_successes_{env_name}_{seed}')
    
    model.set_logger(configure(f'{policy_name}_results_{env_name}_{seed}'))
    model.learn(total_timesteps=int(timesteps), callback=succ_callback, log_interval=4, progress_bar=True)
    model.save(f"{policy_name}_model_{env_name}_{seed}")
    env.save(f'{policy_name}_env_{env_name}_{seed}')

if __name__=="__main__":
    train('myoLegWalk-v0', 'sac', 3e6, 0)

    # Load the trained model
    model = SAC.load("sac_model_myoLegWalk-v0_0")

    # Load the environment
    vec = VecNormalize.load("sac_env_myoLegWalk-v0_0", SubprocVecEnv([make_env('myoLegWalk-v0') for _ in range(16)]))

    env = gym.make('myoLegWalk-v0')
    print(env.sim.model.opt.timestep)
    env.training = False
    env.norm_reward = False

    env.reset()
    frames = []
    done = False
    for i in tqdm.tqdm(range(1000)):
        o = env.get_obs()
        o = vec.normalize_obs(o)
        a, __ = model.predict(o, deterministic=False)
        next_o, r, done, *_, info = env.step(a)
        frame = env.sim.renderer.render_offscreen(
                        width=400,
                        height=400,
                        camera_id=0)
        frames.append(frame)
        if done:
            env.reset()

    filename = 'output.mp4'
    filepath = os.path.join(os.getcwd(), filename)

    fps = 100
    with imageio.get_writer(filepath, fps=fps) as video:
        for frame in frames:
            video.append_data(frame)

    env.close()`

This implementation can reach 1e4 reward within 3e6 timesteps, and produce a reasonable visual results.

https://github.com/user-attachments/assets/15e69773-1f5e-4be7-8a44-38fa6de31eed

Also, the RL-E2E can produce good result on myoLeg-Rough too using the DynSyn E2E implementation.

https://github.com/user-attachments/assets/bfc28bab-0380-4021-bae0-e27787ca2cfb

So I wonder why is the result better than reported?

InckieUltra, 2025-04-10:
> Hello, I'm also curious about this and wonder if you have some average results across seeds (beyond the current `seed=0` result) for reference. Also, based on [SAR tutorial](https://github.com/MyoHub/myosuite/blob/main/docs/source/tutorials/SAR/SAR_tutorial.ipynb), SAR's implementation uses a training frequency: `((1, "episode")` that is different from DynSyn's `((4, "step"))` in their E2E baselines, and that may have caused the difference in their E2E performances.

In fact, I think I made a mistake in the minimal implementation.
In DynSyn, it actually uses gradient_steps=4 instead of ((4, "step")) with gradient_steps=-1, I have edited the code.
Apologize if anyone tries the minimal implementation to verify.

I will provide the results of gradient_steps=4 using seed 0, 42, 114514, and results of gradient_steps=-1 using seed 0 later today.

InckieUltra, 2025-04-11:
[image]
This is the result of using grad_steps=4

[image]
The results of using grad_steps=-1, I dont have time to run multiple experiments with this setup cause it consumes too much time.

And it seems that grad_steps make a difference? It both reduces training time and improves the final results when using stable baselines3.

[image]
I tried to make a similar environment in mjx, and use brax sac to train it. With parameter grad_updates_per_step=32 it also works (See blue curve).

Vittorio-Caggiano, 2025-04-11:
@InckieUltra thanks for the detailed update. Can you share a bit more about the comparisons and the conclusions? Very exciting to see you tested also with mjx/brax. Any insight on the difference in speed?

cc: @jamesheald @Balint-H @ttktjmt @sagarverma

InckieUltra, 2025-04-11:
1. The comparison mainly compares the sac parameter "gradient_steps" in sb3. 

When using gradient_steps=4, in the log, n_updates is 1/4 total_timesteps. And the result can reach 1e4. 
This setup is the same as the minimal implementation I provided above.

When using gradient_steps=-1, in the log, n_updates is almost the same as total_timesteps. The result is very poor.
I ran with change
'''model = SAC('MlpPolicy', env, learning_rate=linear_schedule(.001), buffer_size=int(1e6),
            gradient_steps=4, policy_kwargs=policy_kwargs, verbose=1)''' 
to 
'''model = SAC('MlpPolicy', env, learning_rate=linear_schedule(.001), buffer_size=int(1e6), train_freq=((4, "step")),
            gradient_steps=1, policy_kwargs=policy_kwargs, verbose=1)'''
The "train_freq" seems not important, cause I currently run with train_freq as default for 1e5 total_timesteps, and the result is as poor as  ((4, "step")).

My conclusion is this parameter ("gradient_steps") seems to affect the training result a lot. But I cant explain why.

2. I ran the code in mujoco and mjx both on a computer with 4070tis 16G, AMD Ryzen 5 9600X 6-Core, Ubuntu 24.04.
On CPU mujoco, I use 16 parallel environments, it takes me **1 hour to train 3e6 timesteps**, using gradient_steps=4, and 2.5 hours using gradient_steps=-1.
On MJX, I use 128 parallel environments, about **10 minutes to train 3e6 timesteps**. (grad_updates_per_step=32, buffer_size=1e6, episode_length=1000), and I change the physics timestep to 1/300, the control frequency is still 100 Hz.
Here are two results I get.

https://github.com/user-attachments/assets/84bfb93d-95eb-45cd-b47e-92516ff82daf
https://github.com/user-attachments/assets/b520be2c-ae7e-4b58-9ad1-98db84910691

I cant guarantee the environment I make is exactly the same as myolegwalk, also, my implementation may not be optimized.

Beanpow, 2025-04-24:
In SB3, the policy is updated for a total of $$\frac{\text{total number of samples}}{\text{number of environments} \times \text{train freq}} \times \text{gradient steps} $$ times. (Here, train_freq is a parameter in SB3 that indicates how many environment steps are taken before a policy update. The default value is 5 steps, meaning all environments perform 5 steps before updating.)

Therefore, it can be observed that, under the condition of sampling the same number of samples, the more environments there are, the fewer total updates are performed, which leads to worse final performance. On the other hand, the larger the gradient_steps, the more updates there are, resulting in better final performance, though the training time increases.

In our experiments, setting the number of environments to 1 yielded the best sample efficiency. However, due to the lack of parallelism, the total training time (wall clock time) increased significantly. Thus, to achieve better performance in as little wall clock time as possible, we set the number of environments to 80 (the number of logical cores), and appropriately increased gradient_steps. Although the total time becomes slightly longer, the final performance improves.

closed 2025-04-24 by InckieUltra

InckieUltra, 2025-04-24:
> In SB3, the policy is updated for a total of total number of samples number of environments × train freq × gradient steps times. (Here, train_freq is a parameter in SB3 that indicates how many environment steps are taken before a policy update. The default value is 5 steps, meaning all environments perform 5 steps before updating.)
> 
> Therefore, it can be observed that, under the condition of sampling the same number of samples, the more environments there are, the fewer total updates are performed, which leads to worse final performance. On the other hand, the larger the gradient_steps, the more updates there are, resulting in better final performance, though the training time increases.
> 
> In our experiments, setting the number of environments to 1 yielded the best sample efficiency. However, due to the lack of parallelism, the total training time (wall clock time) increased significantly. Thus, to achieve better performance in as little wall clock time as possible, we set the number of environments to 80 (the number of logical cores), and appropriately increased gradient_steps. Although the total time becomes slightly longer, the final performance improves.

Thanks for your detailed explanation! That does make sense.

sjMOK, 2025-05-15:
Hi, @InckieUltra and everyone! Sorry to comment on a closed issue.

Your issue looks really interesting, and I tried to reproduce the results on my own. While doing so, a couple of questions came up, and I'd really appreciate it if you could help clarify them. Just for context, I'm fairly new to reinforcement learning and musculoskeletal simulation.

**First**, I trained the policy using 16 parallel environments, just like in your code. The average episode reward increased in a similar pattern, which was great to see. However, when I visualized the agent's motion, it showed some very unnatural gait patterns (e.g., walking with one leg only). I understand that this could be due to the large action space.
My question is: was the video you shared the best result among several runs (in terms of natural human-like gait), or did that single run produce such a clean walking behavior?

**Second**, I'm curious about how to run training with different random seeds, like you used the seed of 0, 42 114514 . This may be due to my limited understanding of RL and SB3, and is the code below a right way to do it? If not, could you point out where in the code I should modify the seed? 

```
def make_env(env_id, seed=0):
    def _init():
        env = gym.make(env_id)
        env = Monitor(env)
        env.reset(seed=seed)
        return env
    return _init
```
Sorry for the long question, and I'd really appreciate any guidance!

sjMOK, 2025-05-16:
Thank you @InckieUltra for the kind reply!

Sorry for my poor description about the unnatural motion, and below is the video of it.
It seems that my trained model lacks sufficient rotation in the left knee joint, which is what I meant by “walking with one leg” (my earlier explanation wasn't very clear).

https://github.com/user-attachments/assets/4b4b0c75-4d09-4348-9474-ada741301da2

Can I also ask how you chose the number of environments and the hyperparameters for the SAC model? You mentioned that you referred to the DynSyn paper, but the hyperparameter settings seem a bit different from those used in the paper with 80 environments. I'm interested in how performance is affected by  hyperparameter tuning of RL algorithms!

InckieUltra, 2025-05-17:
> Thank you [@InckieUltra](https://github.com/InckieUltra) for the kind reply!
> 
> Sorry for my poor description about the unnatural motion, and below is the video of it. It seems that my trained model lacks sufficient rotation in the left knee joint, which is what I meant by “walking with one leg” (my earlier explanation wasn't very clear).
> 
>  output.mp4 
> Can I also ask how you chose the number of environments and the hyperparameters for the SAC model? You mentioned that you referred to the DynSyn paper, but the hyperparameter settings seem a bit different from those used in the paper with 80 environments. I'm interested in how performance is affected by hyperparameter tuning of RL algorithms!

The result is similar to what I get. I didnt heavily tune the SAC parameters for this MyoLeg-Walk environment.

If you want more natural gaiting, it's better to use imitation learning instead of designing reward. You can refer to Kinesis (imitation learning in mujoco)https://github.com/amathislab/Kinesis.