MyoHub/myosuite #346: Unsupported environment format in Tutorial 4 'Train_Policy'

abdulmaajidaga, 2025-09-29:
While the running tutorial 4, this error shows up...

Unsupported environment format
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[9], [line 19](vscode-notebook-cell:?execution_count=9&line=19)
     17 seed = 123
     18 rl_step_size = 0.1
---> [19](vscode-notebook-cell:?execution_count=9&line=19) e = GymEnv(env)
     21 policy = MLP(e.spec, hidden_sizes=policy_size, seed=seed, init_log_std=-0.25, min_log_std=-1.0)
     23 baseline = MLPBaseline(e.spec, reg_coef=1e-3, batch_size=64, hidden_sizes=vf_hidden_size, \
     24                     epochs=2, learn_rate=1e-3)

File ~/Documents/mjrl/mjrl/utils/gym_env.py:30, in GymEnv.__init__(self, env, env_kwargs, obs_mask, act_repeat, *args, **kwargs)
     28 else:
     29     print("Unsupported environment format")
---> [30](https://file+.vscode-resource.vscode-cdn.net/home/abdul/Documents/myosuite/docs/source/tutorials/~/Documents/mjrl/mjrl/utils/gym_env.py:30)     raise AttributeError
     32 self.env = env
     33 self.env_id = env.spec.id

AttributeError:

how do I fix it?

Vittorio-Caggiano, 2025-09-30:
> While the running tutorial 4, this error shows up...
> 
> ## Unsupported environment format
> AttributeError Traceback (most recent call last) Cell In[9], line 19 17 seed = 123 18 rl_step_size = 0.1 ---> 19 e = GymEnv(env) 21 policy = MLP(e.spec, hidden_sizes=policy_size, seed=seed, init_log_std=-0.25, min_log_std=-1.0) 23 baseline = MLPBaseline(e.spec, reg_coef=1e-3, batch_size=64, hidden_sizes=vf_hidden_size, 24 epochs=2, learn_rate=1e-3)
> 
> File ~/Documents/mjrl/mjrl/utils/gym_env.py:30, in GymEnv.**init**(self, env, env_kwargs, obs_mask, act_repeat, *args, **kwargs) 28 else: 29 print("Unsupported environment format") ---> [30](https://file+.vscode-resource.vscode-cdn.net/home/abdul/Documents/myosuite/docs/source/tutorials/~/Documents/mjrl/mjrl/utils/gym_env.py:30) raise AttributeError 32 self.env = env 33 self.env_id = env.spec.id
> 
> AttributeError:
> 
> how do I fix it?

Hi @abdulmaajidaga can you provide more details on your setup?
I tried in a colab
```
!pip install myosuite
from myosuite.utils import gym
env = gym.make('myoElbowPose1D6MRandom-v0')
```
end it worked as expected

Vittorio-Caggiano, 2025-09-30:
version of miniconda, operating system, version of myosuite, version of python you are using. can you run `python -m myosuite.tests.test_myo` and share the output of `pip list`?

abdulmaajidaga, 2025-09-30:
I'm running miniconda `25.7.0` on Ubuntu `22.04.5` LTS
myosuite version is `2.11.4 `
python version is `3.9.23`

This is the result of `python -m myosuite.tests.test_myo`
```
(MyoSuite) abdul@abdul-ROG-Zephyrus-G14-GA403UI-GA403UI:~$ python -m myosuite.tests.test_myo
MyoSuite:> Registering Myo Envs
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/gymnasium/envs/registration.py:694: UserWarning: WARN: Overriding environment myoArmReachFixed-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/gymnasium/envs/registration.py:694: UserWarning: WARN: Overriding environment myoSarcArmReachFixed-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/gymnasium/envs/registration.py:694: UserWarning: WARN: Overriding environment myoFatiArmReachFixed-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/gymnasium/envs/registration.py:694: UserWarning: WARN: Overriding environment myoArmReachRandom-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/gymnasium/envs/registration.py:694: UserWarning: WARN: Overriding environment myoSarcArmReachRandom-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/gymnasium/envs/registration.py:694: UserWarning: WARN: Overriding environment myoFatiArmReachRandom-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")

=================================
Testing module::  MyoChallenge Suite
Testing env:  myoChallengeBaodingP1-v1
    MyoSuite: A contact-rich simulation suite for musculoskeletal motor control
        Vittorio Caggiano, Huawei Wang, Guillaume Durandau, Massimo Sartori, Vikash Kumar
        L4DC-2019 | https://sites.google.com/view/myosuite
    
Testing env:  myoChallengeBaodingP2-v1
Testing env:  myoChallengeBimanual-v0
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/myosuite/envs/myo/myochallenge/bimanual_v0.py:434: DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)
  self.manip_joint_range = np.arange(model.joint("manip_object/freejoint").qposadr,
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/myosuite/envs/myo/myochallenge/bimanual_v0.py:437: DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)
  self.manip_dof_range = np.arange(model.joint("manip_object/freejoint").dofadr,
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/myosuite/envs/myo/myochallenge/bimanual_v0.py:172: DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)
  self.obj_vert_addr = np.arange(self.sim.model.mesh(self.obj_mid).vertadr,
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/myosuite/envs/myo/myochallenge/bimanual_v0.py:174: DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)
  q = self.sim.model.geom(self.obj_gid - 1).quat
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/myosuite/envs/myo/myochallenge/bimanual_v0.py:178: DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)
  self.sim.model.geom(self.obj_gid - 1).quat = [1, 0, 0, 0]
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/myosuite/envs/myo/myochallenge/bimanual_v0.py:179: DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)
  self.sim.model.mesh_vert[self.obj_vert_addr] += (self.sim.model.geom(self.obj_gid - 1).pos
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/myosuite/envs/myo/myochallenge/bimanual_v0.py:180: DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)
  - self.sim.model.geom(self.obj_gid).pos)[None, :]
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/myosuite/envs/myo/myochallenge/bimanual_v0.py:182: DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)
[... cut ...]
threadpoolctl            3.6.0
tomli                    2.2.1
torch                    2.8.0
torchvision              0.23.0
tornado                  6.5.1
tqdm                     4.67.1
traitlets                5.14.3
transforms3d             0.4.2
triton                   3.4.0
typing_extensions        4.14.1
tzdata                   2025.2
urllib3                  2.5.0
wandb                    0.15.12
wcwidth                  0.2.13
wheel                    0.45.1
zipp                     3.23.0

```

abdulmaajidaga, 2025-09-30:
This is the result for `python -c "from myosuite.utils import gym; env = gym.make('myoElbowPose1D6MRandom-v0')"`
```
(MyoSuite) abdul@abdul-ROG-Zephyrus-G14-GA403UI-GA403UI:~$ python -c "from myosuite.utils import gym; env = gym.make('myoElbowPose1D6MRandom-v0')"
MyoSuite:> Registering Myo Envs
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/gymnasium/envs/registration.py:694: UserWarning: WARN: Overriding environment myoArmReachFixed-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/gymnasium/envs/registration.py:694: UserWarning: WARN: Overriding environment myoSarcArmReachFixed-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/gymnasium/envs/registration.py:694: UserWarning: WARN: Overriding environment myoFatiArmReachFixed-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/gymnasium/envs/registration.py:694: UserWarning: WARN: Overriding environment myoArmReachRandom-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/gymnasium/envs/registration.py:694: UserWarning: WARN: Overriding environment myoSarcArmReachRandom-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/abdul/miniconda3/envs/MyoSuite/lib/python3.9/site-packages/gymnasium/envs/registration.py:694: UserWarning: WARN: Overriding environment myoFatiArmReachRandom-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
    MyoSuite: A contact-rich simulation suite for musculoskeletal motor control
        Vittorio Caggiano, Huawei Wang, Guillaume Durandau, Massimo Sartori, Vikash Kumar
        L4DC-2019 | https://sites.google.com/view/myosuite

```
Yes, running the tutorial in jupyter notebook
Yes, I'm running the right kernel

abdulmaajidaga, 2025-10-03:
@Vittorio-Caggiano can you share the colab link instead?

abdulmaajidaga, 2025-10-03:
@Vittorio-Caggiano what can I do to fix this?

Vittorio-Caggiano, 2025-10-03:
can you try this and share the output?
``` bash
python3 -m ipykernel install --user --name=<name of your environment>
jupyter nbconvert --to notebook --execute <absolute path of tutorial> --ExecutePreprocessor.kernel_name=<name of your environment>
```

Vittorio-Caggiano, 2025-10-03:
@abdulmaajidaga while I was retesting some of the tutorials I realized there might be a bug and started this PR https://github.com/MyoHub/myosuite/pull/351 . please refer to the code in there if you get stuck testing those tutorials

abdulmaajidaga, 2025-10-04:
@Vittorio-Caggiano sorry for replying late, thank you for sharing the PR!