MyoHub/myosuite #347: matmul: Input operand 1 has a mismatch in Tutorial 6

abdulmaajidaga, 2025-09-30:
After running this piece of code in Jupyter Notebook,
```
model1 = loader.get_sim(None, 'hand/myohand.xml')
tausmooth = 5
model1.actuator_dynprm[:,2] = tausmooth
data1 = mujoco.MjData(model1)
all_ctrl = np.zeros((traj.shape[0], 1+model1.nu))
for idx in tqdm(range(traj.shape[0])):
    target_qpos = traj[idx, 1:]
    qfrc = get_qfrc(model1, data1, target_qpos)
    ctrl = get_ctrl(model1, data1, target_qpos, qfrc, 100, 5)
    data1.ctrl = ctrl
    mujoco.mj_step(model1, data1)
    all_ctrl[idx,:] = np.hstack((data1.time, ctrl))
```
I get this error,
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[32], [line 9](vscode-notebook-cell:?execution_count=32&line=9)
      7 target_qpos = traj[idx, 1:]
      8 qfrc = get_qfrc(model1, data1, target_qpos)
----> [9](vscode-notebook-cell:?execution_count=32&line=9) ctrl = get_ctrl(model1, data1, target_qpos, qfrc, 100, 5)
     10 data1.ctrl = ctrl
     11 mujoco.mj_step(model1, data1)

Cell In[28], [line 33](vscode-notebook-cell:?execution_count=28&line=33)
     31 # ---- ctrl computation
     32 P = 2 * AM.T @ AM
---> [33](vscode-notebook-cell:?execution_count=28&line=33) k = AM @ (gain * act) + AM @ bias - (qfrc / qfrc_scaler)
     34 q = 2 * k @ AM
     35 lb = gain * (1 - act) * ts / (t2 + t1 * (1 - act))

ValueError: matmul: Input operand 1 has a mismatch in its core dimension 0, with gufunc signature (n?,k),(k,m?)->(n?,m?) (size 39 is different from 897)
```

abdulmaajidaga, 2025-09-30:
yes, I receive the same error when I run the code outside Jupyter Notebook

andreh1111, 2025-09-30:
hey, yes sorry @abdulmaajidaga I didn't update the tutorial after the mujoco update (the actuator moment matrix is sparse since a few versions), I'm gonna do a PR now

Vittorio-Caggiano, 2025-10-01:
made a new release with the fix. 
@abdulmaajidaga can you test it and let us know if it works?

closed 2025-10-02 by Vittorio-Caggiano